IvanSolis1989/OpenWrt-DIY #16: ac2100_202009251346 无法联网

bensk2001, 2020-09-25:
你好，我刚刚尝试不保留设置刷了最新的OpenWrt_firmware_xiaomi_redmi-router-ac2100_202009251346，sysupgrade。
结果一直无法联网...我用的电信光猫，IP地址改成了192.168.3.1，网线连接到Redmi AC2100的“Internet”口。
用网线连接，无法自动给电脑分配IP地址，IP地址显示169.254.174.187。用WIFI可以成功连接进入Web管理界面，但是显示无网络连接。

我的路由器是之前刷了别的被编译过的Openwrt，然后在Web界面升级刷入的。尝试过恢复出厂设置，还是不行。
刷别的Openwrt的时候可以正常联网。

不知道这回事啥问题？

今天刚搜到你编译的系统，非常简单稳定，感谢，希望能用上.....

IvanSolis1989, 2020-09-25:
尝试手动吧lan口修改成路由器子网ip 然后设置路由器lan口开启dhcp 强制选项。

bensk2001, 2020-09-26:
> 尝试手动吧lan口修改成路由器子网ip 然后设置路由器lan口开启dhcp 强制选项。

继续求助...我做了如下尝试：

1. "设置路由器lan口开启dhcp 强制选项"——这个勾选然后保存生效，之后路由器的WIFI就挂了，重启也搜不到
2. 光猫物理LAN口分别和路由器物理WAN、LAN口相连，光猫上面有线设备显示为0
2. 插网线连物理LAN口，仍然分配无效IP“169.254.174.187”，默认网关是空着的
3. 手动把网卡的IP地址改为192.168.3.* （这个是我光猫的网段），PING 192.168.3.1 （光猫IP）不通
4. 强制重刷了一遍固件（因为WIFI和网线都连不上），再来过，重复上面的第2和第3，这次手动把电脑网卡IP改为192.168.1.*，PING 192.168.1.1 （路由器IP）不通
5. 我尝试了一周前的那个版本，也是一样的情况

...综上所述，这些是我能想到的尝试了。这个会和固件里面的物理网口驱动有关吗？

bensk2001, 2020-09-26:
我尝试去网上搜了一下，有说DHCP没开的，但至少我用WIFI连接路由器的时候，是可以正常分配IP的。我再重刷然后去看看。 @IvanSolis1989 如果你还想到别的，可以留言给我...

IvanSolis1989, 2020-09-26:
是不是你是光猫拨号所以 软路由的网段不能选择192.168.1.x？尝试设置成10.10.10.10？

bensk2001, 2020-09-26:
> 是不是你是光猫拨号所以 软路由的网段不能选择192.168.1.x？尝试设置成10.10.10.10？

我的光猫不用拨号的，直接可以连接。我用一模一样的方式（红米AC2100 Telnet）刷别的固件是可以正常还是用网线连接光猫和连接电脑。我没把光猫设置到192.168.1.x是因为如果红米的路由器和光猫都在192.168.1.X的网段，路由器连接Internet会非常不稳定，会有冲突。

我下午检查了，DHCP用默认设置，看起来没有什么问题...而且我随便对路由器的lan做任何设置，路由器的WIFI功能直接挂，搜不到热点...5555

IvanSolis1989, 2020-09-27:
会不会是软路由设置的lan口没有绑定 路由器的lan

bensk2001, 2020-09-27:
> 会不会是软路由设置的lan口没有绑定 路由器的lan

我隐约记得物理设置没有问题...让我晚上回去再试试

微信打赏给你留言了~

IvanSolis1989, 2020-09-27:
谢谢啦 已收到

IvanSolis1989, 2020-10-08:
已经调整了config

closed 2020-10-08 by IvanSolis1989