IvanSolis1989/OpenWrt-DIY #34: X86版本的WeTransfer显示失效（Transfer expired）

romejiang, 2020-11-10:
X86版本的WeTransfer一直显示失效。
即便是刚刚编译出来的版本。

[image]

IvanSolis1989, 2020-11-10:
周五编译 只有三天有效期

closed 2020-11-10 by IvanSolis1989

romejiang, 2020-11-13:
五小时前编译的X86，64位的在wetransfer还是显示已失效。
我看了一下今天编译的其他固件都在，猜测，可能是因为X86的固件内容比较大（x86 = 4.55G， 其他的都是1.5G左右）导致wetransfer上传失败或者不接受。

[image]
[image]

romejiang, 2020-11-13:
这是 transfer 项目的说明，是不是 WeTransfer 超过文件大小限制了
[image]