IvanSolis1989/OpenWrt-DIY #46: 建议升级cypress-firmware驱动，修复树莓派4BWIFI稳定性及无法切换高信道的问题

meishimeike, 2020-11-29:
升级cypress-firmware驱动，执行以下Shell即可
sed -i 's/v5.4.18-2020_0402/v5.4.18-2020_0925/g' package/firmware/cypress-firmware/Makefile
sed -i 's/19375-1-53475/21490-1-58574/g' package/firmware/cypress-firmware/Makefile
sed -i 's/firmware\/brcmfmac/firmware\/cyfmac/g' package/firmware/cypress-firmware/Makefile
sed -i 's/b12b0570f462c2f3c26dde98b10235a845a7109037def1e7e51af728bcc1a958/fb71c344e705f5bc9fdae3ce0fbfa299f0af0939ff3ec782aeca0308911d830d/g' package/firmware/cypress-firmware/Makefile

meishimeike, 2020-11-29:
直接粘贴到diy-part后面即可

IvanSolis1989, 2020-11-30:
已添加

IvanSolis1989, 2020-11-30:
采用你的代码编译失败

meishimeike, 2020-11-30:
我的没有失败，基于：https://github.com/P3TERX/Actions-OpenWrt，修改的diy-part2.sh
修改地址：https://github.com/meishimeike/Actions-OpenWrt/blob/main/diy-part2.sh
成功固件地址：https://github.com/meishimeike/Actions-OpenWrt/actions/runs/364997462

meishimeike, 2020-11-30:
应该是这样：
sed -i 's/v5.4.18-2020_0402/v5.4.18-2020_0925/g' package/firmware/cypress-firmware/Makefile
sed -i 's/19375-1-53475/21490-1-58574/g' package/firmware/cypress-firmware/Makefile
sed -i 's/firmware\/brcmfmac/firmware\/cyfmac/g' package/firmware/cypress-firmware/Makefile
sed -i 's/b12b0570f462c2f3c26dde98b10235a845a7109037def1e7e51af728bcc1a958/fb71c344e705f5bc9fdae3ce0fbfa299f0af0939ff3ec782aeca0308911d830d/g' package/firmware/cypress-firmware/Makefile

meishimeike, 2020-11-30:
应该是第三行粘贴的时候，自动給删除了转义符，算github Bug吧。
[image]

IvanSolis1989, 2020-12-01:
好的 大神

IvanSolis1989, 2021-01-07:
SHELL= flock /workdir/openwrt/tmp/.cypress-fmac-v5.4.18-2020_0925.zip.flock -c ' /workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "cypress-fmac-v5.4.18-2020_0925.zip" "fb71c344e705f5bc9fdae3ce0fbfa299f0af0939ff3ec782aeca0308911d830d" "" "https://community.cypress.com/servlet/JiveServlet/download/21490-1-58574/" '

curl -f --connect-timeout 20 --retry 5 --location --insecure https://community.cypress.com/servlet/JiveServlet/download/21490-1-58574/cypress-fmac-v5.4.18-2020_0925.zip
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed

0 0 0 0 0 0 0 0 --:--:-- --:--:-- --:--:-- 0
0 0 0 0 0 0 0 0 --:--:-- --:--:-- --:--:-- 0
curl: (22) The requested URL returned error: 404
Download failed.

curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/cypress-fmac-v5.4.18-2020_0925.zip
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed

0 0 0 0 0 0 0 0 --:--:-- --:--:-- --:--:-- 0
0 0 0 0 0 0 0 0 --:--:-- --:--:-- --:--:-- 0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.

curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/cypress-fmac-v5.4.18-2020_0925.zip
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed

0 0 0 0 0 0 0 0 --:--:-- --:--:-- --:--:-- 0
0 0 0 0 0 0 0 0 --:--:-- --:--:-- --:--:-- 0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.

IvanSolis1989, 2021-01-07:
驱动失效 暂时用源码驱动

meishimeike, 2021-01-08:
官方把下载地址删除了

closed 2021-01-09 by IvanSolis1989

meishimeike, 2021-03-08:
更新一下驱动吧，原来的不稳定
[image]