IvanSolis1989/OpenWrt-DIY #68: SSH connection Always not allowed.

latiefkhan, 2021-01-19:
How tof ix.?

be-engineer, 2021-01-20:
me too。error message is:
Permission denied, please try again.

a00789123, 2021-01-25:
一样样……的问题，不知道是我不懂玩dropbear还是咋滴。

zoraaa, 2021-02-02:
我也一样。。。

配了公钥也是不行。。。

XiaoSong0919, 2021-02-02:
> me too。error message is:
> Permission denied, please try again.

The password is “password”,please try to use it password to login in ssh.

closed 2021-02-18 by IvanSolis1989

azraelrabbit, 2021-03-09:
兄弟,我折腾了一天,才找到这个问题... @XiaoSong0919 
第三个办法: 

修改 `config/linksys-extra.config  ` 文件
增加 一行  `CONFIG_PACKAGE_zsh=y`

重新编译生成固件,也能解决.

azraelrabbit, 2021-03-09:
另外,刚刚看了一下, 
MIPS的所有机型也都会有这个问题,
`config/MIPS/MIPS-extra.config`
文件里也没有默认编译 zsh

azraelrabbit, 2021-03-10:
> > 另外,刚刚看了一下,
> > MIPS的所有机型也都会有这个问题,
> > `config/MIPS/MIPS-extra.config`
> > 文件里也没有默认编译 zsh
> > 其他的都正常.
> 
> https://github.com/IvanSolis1989/OpenWrt-DIY/pull/85/files
> 这样子修改试试

不额外编译zsh还是会有问题的,因为是在 
`scripts/diy-part3.sh`
里面替换了默认ash 到 zsh
依然还是会有找不到zsh 的问题存在. 
第四种办法:
直接去掉 `scripts/diy-part3.sh` 里面的 

```
# Change default shell to zsh
sed -i 's/\/bin\/ash/\/usr\/bin\/zsh/g' package/base-files/files/etc/passwd
```
这两行,其他什么都不要改动.
再编译固件就没问题了.

cxgreat2014, 2021-03-10:
八个小时前发现这一问题，三个小时前自行修改zsh到ash解决，感觉这个问题可以reopen一下

isecret, 2021-08-08:
2021年8月8日，最新版本 newifi d2 的固件问题依然存在。

commit bda88de8 (2021-10-05): Merge pull request #131 from isecret/dev

fix #68 fix MIPS build config zsh shell missing
commit d90eb149 (2021-10-05): fix #68