IvanSolis1989/OpenWrt-DIY #89: Your kernel does not support memory limit capabilities or the cgroup is not mounted. Limitation discarded.

hx210, 2021-02-28:
您的内核不支持内存限制功能或未安装cgroup。 限制已舍弃。 可以往树莓派4b中加入内存限制吗。防止甜糖跑死机，他跑起来就会把内存慢慢的占满

IvanSolis1989, 2021-02-28:
内存限制是什么插件？没见过啊？

XiaoSong0919, 2021-02-28:
> 您的内核不支持内存限制功能或未安装cgroup。 限制已舍弃。 可以往树莓派4b中加入内存限制吗。防止甜糖跑死机，他跑起来就会把内存慢慢的占满

您好，openwrt使用的是linux内核，其cgroup组件并没有被阉割，其目录在/sys/fs/cgroup；如果你想通过cgroup限制进程的使用内存，你可以在/sys/fs/cgroup/memory下面新建一个文件夹，对其进行限制，详细方法就不讲了XD

XiaoSong0919, 2021-02-28:
> 您的内核不支持内存限制功能或未安装cgroup。 限制已舍弃。 可以往树莓派4b中加入内存限制吗。防止甜糖跑死机，他跑起来就会把内存慢慢的占满

Hello, openwrt uses the linux kernel, its cgroup components have not been castrated, and its directory is in /sys/fs/cgroup; if you want to limit the memory used by the process through cgroup, you can go to /sys/fs/cgroup/memory Next, create a new folder and restrict it. The detailed method will not talk about XD

hx210, 2021-03-01:
好的，谢谢

closed 2021-03-01 by hx210